SolarPerformanceInsight/solarperformanceinsight #52: Only allow one array per inverter for new system

alorenzo175, 2020-12-09:
and probably auto-populate the form with one inverter and possibly one array. Eventually we'll support multiple arrays per inverter, so maybe just gray out the Add Array button or alert when adding more than one?

wholmgren, 2021-01-05:
The pvlib functionality for multiple arrays is almost ready to merge into master. One thing that came up in review is that single axis trackers will be restricted to a single array for now. I'm -1 on hacking around that limitation in SPI. It would be far better to improve the situation in https://github.com/pvlib/pvlib-python/issues/1109. In the meantime, we might need to consider graying out that button if a tracker is provided.

wholmgren, 2021-01-15:
set requirements file to use pvlib master for now and remove this limitation in the api. we can also cut pvlib 0.9 pre-releases if that's preferable.

lboeman, 2021-01-22:
After #92 We are at a point where multiple arrays are supported when all arrays have fixed tracking. The dashboard should:

- Only allow selecting between fixed/single axis tracking when one array exists on the inverter.
- Only allow users to add additional arrays (through "Add Array" or "Duplicate Array") if single axis is not selected.

alorenzo175, 2021-02-03:
dashboard limits in #94

closed 2021-02-03 by alorenzo175