SolarPerformanceInsight/solarperformanceinsight #63: options for POA, effective irradiance, module temperature, and cell temperature

wholmgren, 2020-12-12:
pvlib's `ModelChain` provides a handful of methods and data inspection for handling different input data. This issue is meant to summarize the options/behavior and determine requirements/priorities for the Solar Performance Insight development.

| Method | Required | Optional |
| -------- | --------| ----------|
| [`ModelChain.run_model`](https://pvlib-python.readthedocs.io/en/stable/generated/pvlib.modelchain.ModelChain.run_model.html) | `dni, ghi, dhi` | `temp_air, wind_speed, cell_temperature, module_temperature` |
| [`ModelChain.run_model_from_poa`](https://pvlib-python.readthedocs.io/en/stable/generated/pvlib.modelchain.ModelChain.run_model_from_poa.html#pvlib.modelchain.ModelChain.run_model_from_poa) | `poa_global, poa_direct, poa_diffuse` | `temp_air, wind_speed, cell_temperature, module_temperature` |
| [`ModelChain.run_model_from_effective_irradiance`](https://pvlib-python.readthedocs.io/en/stable/generated/pvlib.modelchain.ModelChain.run_model_from_effective_irradiance.html#pvlib.modelchain.ModelChain.run_model_from_effective_irradiance) | `effective_irradiance` | `temp_air, wind_speed, cell_temperature, module_temperature` |

Certainly need to support `run_model`. Likely need to support `run_model_from_poa`. Is `run_model_from_effective_irradiance` of interest to this community or is it more likely to confuse users?

The new ModelChain/PVSystem/Array architecture forces us to use the same ModelChain method and required inputs for all Arrays within an Inverter. We should extend that to the Inverter level in the Insight code. I think the pvlib architecture also  requires the same optional inputs for each array, but this is not clearly documented.

If `temp_air` is not provided, 20 C is assumed. The primary use of `temp_air` is for calculating `module_temperature`, but it's also used in the solar position calculation. If `module_temperature` is provided and `temperature_model=='sapm'`, then `temp_air` is only used in solar position calculation. If `cell_temperature` is provided, then no temperature model is required and again `temp_air` is only used in solar position calculation. If `wind_speed` is not provided, 0 m/s is assumed. The only use of `wind_speed` is in calculating `module_temperature`. It is ignored if `module_temperature` or `cell_temperature` is provided, though this is not clear in the documentation. pvlib handles all of that, so the only challenge for us is communicating the complexity in the data upload user interface.

If only 2/3 of GHI/DNI/DHI are measured, [`ModelChain.complete_irradiance`](https://pvlib-python.readthedocs.io/en/stable/generated/pvlib.modelchain.ModelChain.complete_irradiance.html) can calculate the 3rd component. The results are not checked for physical consistency, but maybe [`pvanalytics.quality.irradiance.check_irradiance_consistency_qcrad`](https://pvanalytics.readthedocs.io/en/latest/generated/pvanalytics.quality.irradiance.check_irradiance_consistency_qcrad.html#pvanalytics.quality.irradiance.check_irradiance_consistency_qcrad) could be used for that. Probably low priority feature.

Do we need to support a GHI-only input? That would require a new ModelChain method and perhaps a model option like `ghi_to_dni_model={'erbs', 'disc', 'dirint', 'dirindex'}`.

cwhanse, 2020-12-14:
`run_model_from_effective_irradiance` applies when the plane-of-array irradiance is measured with a reference cell, rather than a pyranometer. I think both `run_model_from_poa` and `run_model_from_effective_irradiance` are of interest.

I think it would be OK to omit `temp_air` from solar position calculations in Solar Performance Insight. If a user is providing either `module_temperature` (most common) or `cell_temperature` (uncommon) then `temp_air` should not also be needed.

wholmgren, 2020-12-15:
Do you mean to imply that some kind of temperature data should always be required for SPI workflows? I think that's reasonable but it is different from ModelChain.

cwhanse, 2020-12-15:
> Do you mean to imply that some kind of temperature data should always be required for SPI workflows? I think that's reasonable but it is different from ModelChain.

I can't think of a reasonable use of the ModelChain default temperatures, which I don't think were a good idea for pvlib but understand why they were used.

wholmgren, 2020-12-15:
Ok, SPI will require a temperature input. What about wind speed?

> which I don't think were a good idea for pvlib but understand why they were used.

I'd be open to changing that in pvlib.

cwhanse, 2020-12-15:
> Ok, SPI will require a temperature input. What about wind speed?

Same. We could provide a default for both in the web form to save users some typing (20C, 1m/s) - those are the NOCT weather conditions. I think the user should have to explicitly decide which values to use.

wholmgren, 2020-12-15:
I'll leave to @alorenzo175 and @lboeman to decide on feasibility of a web form option for scalar input but I'm guessing that will be relatively difficult to integrate with the API.

cwhanse, 2020-12-15:
Ok without defaults, easy enough for a user to make a Series of 20C.

wholmgren, 2021-02-10:
@alorenzo175 we can close this, right?

alorenzo175, 2021-02-10:
yes

closed 2021-02-10 by alorenzo175