SolarPerformanceInsight/solarperformanceinsight #64: interval label, type requirements

wholmgren, 2020-12-12:
We need to decide if/how we will account for interval label and interval value type. From [solar forecast arbiter definitions](https://solarforecastarbiter.org/definitions/#forecastattrs)

* Interval length - The length of time that each data point represents, e.g. 5 minutes, 1 hour.
* Interval label - Indicates if a time labels the beginning or the ending of an interval average, or indicates an instantaneous value, e.g. beginning, ending, instant
* Interval value type - The type of the data in each interval of the forecast, e.g. mean, maximum, median.

Do we have restrictions on interval length? Longest interval of 1 hour? Is there a need to support or restrict intervals shorter than 1 minute? We'll have an upper limit on upload data size, so shorter intervals may run into restrictions on total length.

Can we require a particular interval label? I doubt it. Can we treat them all as instantaneous? Probably fine for 5 - 15 minute intervals, but we might need to do better for 1 hour intervals. What's the scope for 1.0 vs someday? If we perform upsampling and averaging, what's the scope for Insight vs. pvlib? Relevant discussion in the [pvlib google group](https://groups.google.com/g/pvlib-python/c/b4uMI3v3qOc).

Interval value type. Only mean or instantaneous. I don't see an application for anything else.

Also related: we should require that all time series data has the same properties.

cwhanse, 2020-12-14:
The great majority of irradiance data will be time-averaged values. I think there are three issues that should be addressed in the first release:
- interval labelling, left or right endpoint
- interval closure e.g., is the last hour of a day 23:00-23:59 or 23:01-24:00?
- pairing solar position with each interval. I have a preference here to use solar position at the interval midpoint. We could apply the midpoint rule for intervals e.g. 30 minutes or longer, but it might be simpler to always use the midpoint.

wholmgren, 2020-12-14:
Is there a realistic situation in which labeling and closure do *not* go together for the expected users? 

> I have a preference here to use solar position at the interval midpoint. 

Do you think we would implement this in the SPI code or the `ModelChain` code? SPI might be faster but `ModelChain` would be better in the long run. If in the SPI code, I think we'd add/subtract half of the interval length from the datetime index before sending into `ModelChain`, then add it back to the relevant `ModelChainResults` attribute indices before reporting them out to the user. This does assume that the user has regular intervals, which might present a challenge to some users.

> it might be simpler to always use the midpoint.

Yes, likely true if we can assume time averaged input.

cwhanse, 2020-12-14:
> Is there a realistic situation in which labeling and closure do _not_ go together for the expected users?

I doubt it. I'm inclined to push the burden of adjusting data onto the user.

> > I have a preference here to use solar position at the interval midpoint.
> 
> SPI might be faster but `ModelChain` would be better in the long run. 

Agree. At this stage speed of delivery is the priority, I think.

> If in the SPI code, I think we'd add/subtract half of the interval length from the datetime index before sending into `ModelChain`, then add it back to the relevant `ModelChainResults` attribute indices before reporting them out to the user. 

Sounds ok to me.

> This does assume that the user has regular intervals, which might present a challenge to some users.

Good point. I suppose we could compute a Series of interval lengths etc. don't know how that would affect performance.

alorenzo175, 2020-12-22:
In https://github.com/SolarPerformanceInsight/solarperformanceinsight/pull/70/files#diff-61dbb2d4dbcbefde65fbf57cb25291a4cd8d3316b72702d6c950a7e618be7c70R557-R579, I've assumed that a user will provide some information about the intervals that they expect to upload. This is makes it more straightforward to validate the data upload and to calculate midpoints for solar position. I imagine that this specification might go on the page in the dashboard before you even start uploading any data, although I guess it could also go on the upload page.

From the call yesterday, it sounds like it'd be helpful to allow missing periods if possible. I think we can accomplish that by requiring the time index information and then ensuring any uploaded data can be reindexed to that index with nans filled into the appropriate slots and just running the calculations. We should probably also check something simple like `len(nans_in_data) <= 0.1 * len(whole_series)` to catch instances where one, e.g., specifies that you have 15min intervals but upload data with 1hr intervals.

Thoughts?

wholmgren, 2020-12-22:
> I've assumed that a user will provide some information about the intervals that they expect to upload... I imagine that this specification might go on the page in the dashboard before you even start uploading any data, although I guess it could also go on the upload page.

Since the dashboard is reading the first few lines of the file, how about letting it try to infer the interval length and fill in the field for the user? Upload page could include a warning if the first couple of intervals are inconsistent with the value the user provides.

alorenzo175, 2020-12-22:
Sure, that's fine so long as the form is provided for them to fill/tweak. I would put inferring the interval and start/end from the file as a low priority though.

PR #80 (2021-01-11): Validate job data uploads
closes #64 closes #77

Validates uploaded job data to: 
- conform to the times as specified for the job including adding nans if appropriate
- fail if more than 10% of times are missing
- fail if more than 10% of times are extra (compared to time index spec for job)
- fail if any columns are missing
- only save required columns
- save as arrow file format to database
- accepts either arrow file format data or csv
- validates times and numeric columns

Returns some information about the values that were uploaded like how many were nan or how many and which times were m [... cut]
closed 2021-01-11 by alorenzo175