SolarPerformanceInsight/solarperformanceinsight #65: interface for starting analysis workflow

wholmgren, 2020-12-12:
How are we going to present various analyses pathways to the user? A few options:

A. Select workflow, then select site, then upload data. User clicks buttons for one of **Calculate predicted performance**, **Calculate expected performance**, etc. Next page shows the table of sites with previously defined metadata, and user clicks one of these names, clicks "create and save a new site", or uploads a json file. For selection of an existing site, next page shows the relevant site layout with buttons for uploading time series data and associating data with the right objects. For selection of create a new site, go to standard site creation page, but when user clicks "save site" they go directly to the relevant time series data input page. For "uploads a json file", API validates the input and then goes directly to the time series data input page. Pros: easier for new users. Implies API routes like `/workflow1` that naturally accept the metadata and data in one post. Keeps the metadata management very distinct from the workflows. Cons: more clicking. 

B. Select site, then select workflow, then upload data. User sees the table of sites with previously defined metadata and selects one of these or clicks "create a new site". Click on existing site shows the detailed metadata, with buttons **Edit metadata**, **Calculate predicted performance**.... Same time series data input pages follow. Pros: ? Cons: site page becomes even more complicated. Implies API routes like `/site1/workflow1` and that's inconsistent with the idea that you'd simultaneously post metadata and data.

C. Select workflow for a particular site, then upload data. User sees the table of sites with previously defined metadata, and there are buttons for **Calculate predicted performance**, etc. next to each name. Pros: all options available from a single page. Cons: requires that user understands the entire workflow before doing anything. To me, this also implies API routes like `/site1/workflow1`.

Also worth revisiting the use cases with a focus on the time series data:

- [1.A. Calculate predicted performance](https://solarperformanceinsight.org/usecases/#uc1A). One upload: 
  - original weather data

- [1.B. Calculate expected performance](https://solarperformanceinsight.org/usecases/#uc1B). One upload: 
  - actual weather data. Data quality checks may be applied if implemented (#62).

- [1.C Account for snow and soiling](https://solarperformanceinsight.org/usecases/#uc1C). I don't think we need a special workflow for this (also these models still need to be hooked into ModelChain).

- [2.A Compare predicted and actual performance](https://solarperformanceinsight.org/usecases/#uc2A). Three uploads: 
  - original weather data, 
  - actual weather data, 
  - actual performance. We're not saving model results, so "selects results of a Predicted Performance run 1.A." is no longer relevant.

- [2.B Compare predicted and expected performance](https://solarperformanceinsight.org/usecases/#uc2B). Three uploads: 
   - original weather data, 
   - actual weather data, 
   - predicted performance

- [2.C Compare expected and actual performance](https://solarperformanceinsight.org/usecases/#uc2C). Two uploads: 
  - actual weather data, 
  - actual performance data

- [2.D Calculate weather-adjusted performance ratio](https://solarperformanceinsight.org/usecases/#uc2D). Two uploads:
   - actual performance data, and 
  - either modeled performance or actual weather data.
 
The data upload pages will look essentially the same for 1A/1B and 2A/2B, so we might consider reducing the options to "Calculate predicted or expected performance", "Compare predicted and actual or expected performance", "Compare expected and actual performance", and "Calculate weather adjusted performance ratio". But if the reports for 1A and 1B need to be different (even with just slightly different text) then we should keep the workflows separate too.

ncroft-b4, 2020-12-14:
Leaning option 1. Less interface to manage, easier to get directly to the result when starting with a new site, and it seems to be an easier pathway from that to where multiple sites might be selected for the same workflow for a portfolio level analysis at once.

lboeman, 2020-12-14:
After team discussion it seems we are leaning toward C. I'll work on a mock up with the following characteristics
System management page with rows containing
- Site name
- Edit button
- Delete button
- Predict performance button. Leads to another page for usecases 1A and 1B.
- Compare performance button. Leads to another page supporting 2A, 2B, and 2C.
- Calculate PR button. Leads to page for 2D.

alorenzo175, 2020-12-17:
> * [2.A Compare predicted and actual performance](https://solarperformanceinsight.org/usecases/#uc2A). Three uploads:
>
>       * original weather data,
>       * actual weather data,
>       * actual [...]

So we will always calculate predicted performance from the original weather data? Do we want the option to upload the predicted performance data? Similar for 2B but option is to calculate instead of providing predicted performance.

>     * [2.D Calculate weather-adjusted performance ratio](https://solarperformanceinsight.org/usecases/#uc2D). Two uploads:
>       
>       * actual performance data, and
>       * either modeled performance or actual weather data.

I don't understand this. 1. You always need to provide the actual weather data to "account for irradiance and temperature", no? 2. Is "modeled performance" what is called "expected performance" elsewhere? 3. How is this weather adjusment different from the adjustment in 2A and B?

cwhanse, 2020-12-17:
> So we will always calculate predicted performance from the original weather data? 

Yes, the definition of "predicted performance" is the output of a model using the weather and model settings locked-down at the time a calculation was done for financing.

> Do we want the option to upload the predicted performance data? Similar for 2B but option is to calculate instead of providing predicted performance.

Probably a desired feature to upload archived performance data. I'm not clear on the priority of this feature.
 
> I don't understand this.

"Weather-adjusted" performance ratio is a single scalar value that summarizes how measured data compares to a model. I'll find a reference for the calculation. In concept,

PR = f(time series of actual performance, time series of expected performance).

In the 2.D description "modeled performance" = expected performance.

Hope this helps.

alorenzo175, 2020-12-18:
> Probably a desired feature to upload archived performance data. I'm not clear on the priority of this feature.

OK, for now I'll leave out the option, but it should be relatively easy to add later.

>    "Weather-adjusted" performance ratio is a single scalar value that summarizes how measured data compares to a model. I'll find a reference for the calculation. In concept,
> 
> PR = f(time series of actual performance, time series of expected performance).
> 
> The time series of expected performance is either uploaded, or is calculated from uploaded weather by the SPI.
> 
> In the Use Case 2.D description "modeled performance" = expected performance.

The pvanalytics function https://github.com/pvlib/pvanalytics/blob/master/pvanalytics/metrics.py helps. Is this how we will calculate the weather-adjusted PR for SPI (possibly switching out the cell temperature model depending on how the system is defined)? 

I guess that raises another question. I've been assuming any provided performance timeseries is of AC power. Is that a valid assumption?

cwhanse, 2020-12-18:
Is this how we will calculate the weather-adjusted PR for SPI (possibly switching out the cell temperature model depending on how the system is defined)?

Yes in principle. For implementation I'll factor the PR calculation out of that function so SPI doesn't have to monkey with the models.

> I guess that raises another question. I've been assuming any provided performance timeseries is of AC power. Is that a valid assumption?

Yes

cwhanse, 2020-12-18:
@alorenzo175 feedback on this refactor of that pvanalytics function? SPI would use the two helpers and pvlib.pvsystem.pvwatts_dc. May not be worth importing the pvanalytics helpers but I think in the long run SPI shouldn't have to maintain it's own implementation.

`gamma_pdc` is a bit of an issue. It is an input for the PVWatts model but not for PVsyst. (For CEC its not in the pvlib database, this isn't `gamma_ref`).  I think it should be a new user input for the PVsyst (and CEC models). It's always on a module datasheet.

For multiple arrays, `gamma_pdc` should be the DC power weighted average over the arrays.

```
#  irradiance is plane of array, either poa_global or effective_irradiance
tavg = _calc_cell_temp_weighted(cell_temperature, irradiance) 
# pdc0 is total DC capacity in all arrays
# gamma_pdc is the temperature coefficient of DC power
modeled = pvlib.pvsystem.pvwatts_dc(irradiance, cell_temperature, pdc0, gamma_pdc, temp_ref=tavg)
performance_ratio = _performance_ratio(measured, modeled)
```

```
def _performance_ratio(measured, modeled):
    return measured.sum() / modeled.sum()

def _calc_cell_temp_weighted(cell_temperature, irradiance):
    numerator = cell_temperature * irradiance
    return numerator.sum() / irradiance.sum()

def performance_ratio_nrel(poa_global, temp_air, wind_speed, pac, pdc0,
                           a=-3.56, b=-0.075, deltaT=3, gamma_pdc=-0.00433):
```

alorenzo175, 2020-12-18:
At the moment we plan to allow users to upload weather on a per array basis. In this case, should we calculate `modeled` for each array individually and then add the timeseries together before calling `_performance_ratio` or calculate an average cell_temp and irradiance? 

If we were to calculate an average cell temp and irradiance, I think a simpler refactor for SPI might be to only have to call a single function from pvanalytics like

```python
def _performance_ratio_from_cell_temp(measured, irradiance, cell_temperature, pdc0, gamma_pdc):
    tavg = _calc_cell_temp_weighted(cell_temperature, irradiance)
    modeled = pvlib.pvsystem.pvwatts_dc(irradiance, cell_temperature, pdc0, gamma_pdc, temp_ref=tavg)
   return _performance_ratio(measured, modeled)

def performance_ratio_nrel(*):
    cell_temperature = sapm_cell(...)
    return _performance_ratio_from_cell_temp(...)
```

cwhanse, 2020-12-21:
> At the moment we plan to allow users to upload weather on a per array basis. In this case, should we calculate `modeled` for each array individually and then add the timeseries together before calling `_performance_ratio`

Yes, I think it is safer to calculated `modeled` (basically, DC output) per array than to calculate it once by averaging cell_temp and irradiance. At some point the performance ratio calculation might account for other factors such as inverter clipping or efficiency.

By doing the per-array calculation, `gamma_pdc` is also per array and the averaging I described above isn't needed.

wholmgren, 2021-01-12:
@alorenzo175 @cwhanse @ncroft-b4 here's my attempt to map the use cases to python functions. 

Use Case | Uploads | Functions
----------|--------- |-----------
[1.A. Calculate predicted performance](https://solarperformanceinsight.org/usecases/#uc1A) | <ul><li>original weather data </li></ul> | <ul><li>`ModelChain.run_model(original_weather)`</li></ul>
[1.B. Calculate expected performance](https://solarperformanceinsight.org/usecases/#uc1B) | <ul><li>actual weather data. </li></ul> | <ul><li>`ModelChain.run_model(actual_weather)`</li></ul>
[2.A Compare predicted and actual performance](https://solarperformanceinsight.org/usecases/#uc2A) | <ul><li>original weather data</li><li>actual weather data</li><li>actual performance</li></ul> | <ul><li>New SPI/pvanalytics/pvlib function to adjust actual performance for differences in actual weather and original weather</li></ul>
[2.B Compare predicted and expected performance](https://solarperformanceinsight.org/usecases/#uc2B) | <ul><li>original weather data</li><li>actual weather data</li><li>predicted performance</li></ul> | <ul><li>`ModelChain.run_model(actual_weather)` </li> <li>New SPI/pvanalytics/pvlib function to adjust expected performance for differences in actual weather and original weather</li> </ul>
[2.C Compare expected and actual performance](https://solarperformanceinsight.org/usecases/#uc2C) | <ul><li>actual weather data</li><li>actual performance data</li></ul> | <ul><li>`ModelChain.run_model(actual_weather)` </li><li> stats on `ModelChainResults` and actual performance</li></ul>
[2.D Calculate weather-adjusted performance ratio](https://solarperformanceinsight.org/usecases/#uc2D) | <ul><li>actual performance data</li><li>either modeled performance or actual weather data</li></ul> | <ul><li>? </li><li> `pvanalytics.metrics.performance_ratio_nrel` </li><li> ?</li></ul>

lboeman, 2021-01-12:
> * New SPI/pvanalytics/pvlib function to adjust actual performance for differences in actual weather and original weather

I was under the impression that the initial implementation would make these comparisons by using the output of `ModelChain.run_model(original_weather)` and `ModelChain.run_model(actual_weather)` (or some permutation thereof). And that the comparison using this adjustment was being kicked into a new use case. Is that right?

I ask because of a point brought up by @cwhanse and @ncroft-b4: In usecase 2.B. if we were to run the model twice we might want to provide users with the prompts for type of irradiance, granularity, and how to calculate cell temperature for both runs (since this is effectively comparing the outputs of 1A and 1B).

It seems like a reasonable change, but doesn't fit into the api workflow we've created so far. Currently we assume that all the provided weather data will be processed using the same type of irradiance, granularity, and temperature type.

ncroft-b4, 2021-01-12:
Can we add a 4th column to track what the output looks like also?

Use Case | Uploads | Functions | Basic Output
-- | -- | -- | --
1.A. Calculate predicted performance | original weather data | ModelChain.run_model(original_weather) | Time Series, Predicted Power
1.B. Calculate expected performance | actual weather data. | ModelChain.run_model(actual_weather) | Time Series, Expected Power
2.A Compare predicted and actual performance | original weather data, actual weather data, actual performance | New SPI/pvanalytics/pvlib function to adjust actual performance for differences in actual weather and original weather | Percentage Comparison, binned (monthly?) of actual to **weather adjusted predicted**?
2.B Compare predicted and expected performance | original weather data actual weather data predicted performance | ModelChain.run_model(actual_weather)New SPI/pvanalytics/pvlib function to adjust expected performance for differences in actual weather and original weather | Percentage Comparison, binned (monthly? user choice?) of predicted to model expected?
2.C Compare expected and actual performance | actual weather data actual performance data | ModelChain.run_model(actual_weather)stats on ModelChainResults and actual performance | Percentage Comparison, binned (monthly? user choice?) of actual to model expected?
2.D Calculate weather-adjusted performance ratio | actual performance data either modeled performance or actual weather data | pvanalytics.metrics.performance_ratio_nrel | From Use Case: "ratio of actual to modeled performance." Binned, percentage comparison?

I think with 2B, we need to sort out what the goal of that is. If you just take the output of 1A and 1B and directly compare them, it helps answer the question: How different is the performance of my system from my prediction due to weather? If you take 1A, _adjust it for the actual weather_ and compare that to 1B, you might get some help with the question: what is the performance difference from the **model** used for calculating expected vs calculating predicted. 

The second question does not seem to have very much utility to me. It would be a specific use case where you have predicted performance from a different model than pvlib, have compared that to actual using 2A, and want some insight into if part of the delta shown by 2A is due to the pvlib model itself vs the original model. Does that track? If so, I propose we change 2B to remove the "adjusts expected performance to the irradiance and temperature of the predicted performance" and just plan to have it compare the power outputs of predicted and expected.

> In usecase 2.B. if we were to run the model twice we might want to provide users with the prompts for type of irradiance, granularity, and how to calculate cell temperature for both runs (since this is effectively comparing the outputs of 1A and 1B).

Yes, I think we need to be able to support two different ways to process original weather and actual weather for the two different  runs. Could we loop the workflow a bit differently? What if a user can only start from either 1A or 1B, (unless they have their own predicted time series). Then after calculating either predicted or expected, it asks, "do you want to now compare to..." and then offer 2A, 2B, or 2C. If they choose 2B, it loops back into the jobs for 1A or 1B, and if they choose 2A or 2C, it asks for actual performance upload at that point and then goes to the comparison output. It redirects the workflow but would hopefully keep all the underlying pages the same.

cwhanse, 2021-01-12:
1.A - agree with @wholmgren 
1.B - agree with @wholmgren 

For the rest, reading this discussion suggests (to me) that the key question is _how_ to compare two time series, rather than which time series are being compared.

Options:
- bin data (e.g. monthly) and compare summary statistics (e.g. kWhr)
- adjust one time series to the weather conditions of the other time series. The correction is an operation on the time series but the outcome is a single value, akin to a performance ratio. Could be done on binned data. Doesn't make sense to return a time series with the same index as the input.
- ?

wholmgren, 2021-01-12:
For 2B I didn't actually think we were running the model twice. We'd run the model once and make adjustments to the output based on some function of the actual and expected weather (e.g. monthly average GHI).

>Can we add a 4th column to track what the output looks like also?

Sure, I'll edit my post to include your lists so that you don't have to deal with the hassle of mixed markdown and html

> Could we loop the workflow a bit differently? What if a user can only start from either 1A or 1B, (unless they have their own predicted time series). Then after calculating either predicted or [...]

Probably out of scope. It's pretty different from how we understood the requirements. It's not yet implemented, but we do intend to provide options to save the mapping of csv headers to variables, so that should make it relatively quick to repeat something if needed.

ncroft-b4, 2021-01-12:
> For 2B I didn't actually think we were running the model twice. We'd run the model once and make adjustments to the output based on some function of the actual and expected weather (e.g. monthly average GHI).

The use case does say that an option is "selects results of a predicted Performance run 1.A" so I don't think it matters all that much if we make it clear that has to be a separate step and that calculation will not be a part of the 2.B workflow or 1A could be triggered as part of 2B, in the end it is still a model run with original weather vs a model run with actual weather. I still do not see the utility of doing any further correction on that based on differences in the weather, as that might already be the only difference in the datasets. Unless the use case there really is to compare two different models, but then why wouldn't you use the same weather on the different models?

> Probably out of scope. It's pretty different from how we understood the requirements. It's not yet implemented, but we do intend to provide options to save the mapping of csv headers to variables, so that should make it relatively quick to repeat something if needed.

Makes sense. If we pair down 2B to be explicit that the predicted performance time series must be uploaded and will not be calculated, remove the requirement for an upload of original weather, and make 2B just a comparison of calculated expected power vs uploaded predicted power, that simplifies it a lot. The inputs for 2B become just actual weather and predicted performance. If similarly, we remove the calculation of predicted performance from 2A and say that predicted performance has to be run in 1A and then reuploaded in 2A, the inputs for 2A become just actual performance and actual weather, and then we do not have any cases where two weather sets get uploaded at the same time. That bypasses the trouble of having to account for two differently formatted weather sets, I believe.

cwhanse, 2021-01-13:
I regret that we kept the terms "predicted, expected" in the use case descriptions. I think these distinctions have only obscured the intent and made designing the interface difficult. In hindsight, the terms **actual**, **modeled** (in place of expected) and **reference** (in place of predicted) would have been more clear, at least to me.

Regarding the Calculate Performance use cases, the only distinction between 1A and 1B is the weather data which the user uploads each time a calculation is done. When we made the distinction between 1A and 1B we may have been thinking that SPI would retain the original weather/power time series as part of the metadata attached to a system, as a convenience for the user. Since SPI is not retaining that data, from SPI's perspective 1A and 1B are only distinguished by the data the user chooses. I don't think we need two different variations on Calculate Performance.

The Compare use cases.

It is my opinion that 2A should return two performance ratios, one for each of the two input sets (reference and actual). Some may prefer a single value returned (ratio of performance ratios) but that makes me shudder. I would recommend using the NREL weather corrected performance ratio when temperature is available in the input weather, and a simple performance ratio in the rare case where temperature is not available (that capability can be deferred).  For each performance ratio, the input is weather and power data. So there are four uploads needed: actual weather and power, and reference weather and power. There isn't a reasonable basis of comparison if either half of the reference set is missing.

2B should also returns two performance ratios, but the difference from 2A is that only 3 inputs are requested, since modeled power substitutes for actual power (modeled power is calculated by SPI from the actual weather uploaded by the user).

2C is a fundamentally different use case. Here, the opportunity exists to line up the two time series of power (actual and modeled) and to learn something from their differences. It is terribly tempting to work at this but at this point in the project I don't think we have the time or budget. So I suggest that we keep 2C in the interface to provide a point of departure for future development, and for now return two performance ratios.

Since each Compare use case has four time series (two each of weather and power), we can also return some general statistics for each pair): monthly total energy (electric) and insolation, would be my suggestion.

ncroft-b4, 2021-01-13:
I am worried that with that change to 2A and without 2C, a lot of the utility of the tool may be removed. 

With 2A, owners in my experience want to see actual production against the reference adjusted for weather, and like to see the delta in units of kWh. I am not sure many would want to see two performance ratios, unless it is indeed the ratio of the two. (A metric about what percent the system underperformed if usually valuable). A possibility I saw for 2A was to output something such as this table:

Month (or other bin) | Actual (kWh) | Weather Adjusted Reference (kWh) | Difference (kWh) | Percent Actual to WA Reference 
--|--|--|--|--
Jan | ... | ... | ... | ... 
...| ... | ... | ... | ...
Dec| ... | ... | ... | ...
Total| ... | ... | ... | ...

We could also include performance ratios as additional columns, (which starts getting into covering the use case in 2D).

For 2C, could it just be basically a wrapper on 1B, that just includes the actual in the output and provides a table. Then in the future some other useful stats or operations could be added to that. 

Month (or other bin) |Actual (kWh) | Modeled (kWh) | Difference (kWh) | Percent Actual to Modeled 
--|--|--|--|--
Jan | ... | ... | ... | ... 
...| ... | ... | ... | ...
Dec| ... | ... | ... | ...
Total| ... | ... | ... | ...

I do like the idea of using a performance ratio for 2B as the comparison of reference and modeled. Maybe the output could be along the lines of:

Month (or other step) |Modeled (kWh) | Reference (kWh) | Model PR | Reference PR 
--|--|--|--|--
Jan | ... | ... | ... | ... 
...| ... | ... | ... | ...
Dec| ... | ... | ... | ...
Total| ... | ... | ... | ...

cwhanse, 2021-01-13:
> the reference adjusted for weather,

how is this adjustment done?

ncroft-b4, 2021-01-13:
> > the reference adjusted for weather,
> 
> how is this adjustment done?

Are we going to require that users setup a system and its metadata before running 2A? Or is a workflow available where a user does not configure a system but can upload a file that includes reference power, reference ghi/poa, and reference ambient temp along with actual power, measured ghi/poa, and measured ambient or module temp? 

If system configuration is not required, then we need to come up with a calculation that would compare the reference irradaince and temperature to the actual irradiance and temperature and adjusts the reference based on that. I would not like to have this be an option I think. You would need to ask for some system metadata anyway (temp coefficient, orientation at least) and it's unclear how the adjustment should be made, though we could come up with some method. However, one thing this would potentially allow is for a user to only have monthly values for the reference conditions vs having hourly values, if our adjustment method could be done monthly. 

If a system configuration is required, then I think there are two cases: A user has a reference dataset from some other source like PVsyst, or they have/want one from SPI. If they have one from a different source, then one option would be to under the hood run the model twice, once with the reference weather and once with the actual weather, and get the ratio of the outputs and adjust the reference power from the other source based on that. If alternatively the reference is calculated by SPI, then really the weather adjusted reference should just be the modeled - or the model run with actual weather - directly, shouldn't it? This case limits us to accepting the reference in only hourly (or smaller) timesteps.

wholmgren, 2021-01-14:
> I don't think we need two different variations on Calculate Performance.

We considered this early on in this discussion (see first post above and discussed on calls around that time). We decided to keep both options for the initial work in case there was a need to report different timeseries/metrics and under the thinking that it would be easier to simplify than to make more complicated if needed. We're not committed to keeping the distinction.

> Are we going to require that users setup a system and its metadata before running 2A? 

Yes.

cwhanse, 2021-01-15:
> With 2A, owners in my experience want to see actual production against the reference adjusted for weather, and like to see the delta in units of kWh. 

I tinkered with various formulas to adjust reference (original) production to the actual conditions. Any method requires the reference (original) weather.

Case 1 - inverter clipping is minor factor and is neglected.

Case 1.a. data are e.g. monthly. Here, I think we could adjust the original kWhr production by the ratio of insolation: kWhr (adjusted) = kWhr (original) x (kW-hr / m2 (POA irradiance) actual) / (kW-hr / m2 (POA irradiance) original).  This requires only the totals for each inverter e.g. monthly.

Case 1.b. if hourly weather data are available, the adjustment could be modified to include temperature along the lines of NREL's PR method. I'm guessing that both 1a and 1b are of interest.

Case 2 - inverter clipping is significant enough to care about it.

In this case, hourly original weather and power are pretty much required, as is some kind of performance model. The problem is that we can't know from the original weather and power how much was lost from the original power due to clipping. I don't yet see a reasonable path except to pass the original weather through a performance model.

cwhanse, 2021-01-20:
@ncroft-b4 @wholmgren maybe we are close to consensus here.

@ncroft-b4 has proposed some templates for the output of cases 2A-2C. I like them. For the Weather-Adjusted Reference, I think we could use the formula I proposed for Case 1b, even in the case of inverter clipping.

wholmgren, 2021-01-20:
> Case 1.a. data are e.g. monthly.

I thought we decided that we'd only accept hourly or better data.

cwhanse, 2021-01-20:
> > Case 1.a. data are e.g. monthly.
> 
> I thought we decided that we'd only accept hourly or better data.

Yes for measured weather, but also required for reference? If yes, I'm OK with that for v1.0 of this tool.

wholmgren, 2021-01-20:
I don't know how to model PV output using monthly data and I'm not aware of functions in pvlib of pvanalytics that would do it. If you can provide a function that will do what's needed then we can consider it.

cwhanse, 2021-01-20:
I wasn't clear. I think we _could_ in theory accept monthly reference data (energy and insolation) which wouldn't be used as input to a performance model, only as the reference point for cases 2A and 2B.

wholmgren, 2021-01-20:
Ah, that makes sense now and I agree that's useful functionality. It would require some changes to the user interface (@lboeman), and maybe also the validation jobs (@alorenzo175).

wholmgren, 2021-01-21:
After some internal discussion, I think it's best if we get to a prototype assuming that all timestamps in all inputs are the same. That also implies that timestamps for a reference file are the same year as the actuals. I understand that is not ideal  long term behavior, but I think we need to get moving with what we have.

With these constraints, can we develop some pseudocode or even code for use cases 2? It's still not clear to me.

We hope to be returning results for use cases 1 by mid/late next week and perhaps that will help everyone understand what's needed for 2.

alorenzo175, 2021-01-27:
After the call Monday, it's clear that some data restrictions will need to move levels to support 2A and 2B. We had discussed how to make the dashboard form more simple for the compare performance use cases. I think the restrictions we have on the data are important in ensuring that we can proceed with the computation without any gotchas or incompatibilities that only crop up once you try and run the comparison. I think we can support 2A/B while still avoiding those issues by making a form like 
[image]

Then we would be able to support different weather formats for the two uploads, and different time indices. We would add some validation to make sure the predicted time index could be shifted to the actual/expected time index.

wholmgren, 2021-01-27:
That seems pretty reasonable to me. Without yet committing to supporting it, how would we modify this to support uploads of monthly data? Allow step to be specified in other units (like in the Arbiter)? A different button dedicated to switching between time series and monthly? And do we require monthly sums or allow both monthly sums and monthly averages?

alorenzo175, 2021-01-27:
> That seems pretty reasonable to me. Without yet committing to supporting it, how would we modify this to support uploads of monthly data? Allow step to be specified in other units (like in the Arbiter)? A different button dedicated to switching between time series and monthly? 

Button

> And do we require monthly sums or allow both monthly sums and monthly averages?

Seems like averages aren't really used, but I guess either is fine

wholmgren, 2021-03-17:
I think this issue has gone as far it can and we've completed its nominal scope, so I am going to close it. Worthwhile for everyone to review the discussion above and make spin off issues for anything that might have slipped through the cracks.

closed 2021-03-17 by wholmgren