SolarPerformanceInsight/solarperformanceinsight #74: report design

wholmgren, 2020-12-17:
What will the reports look like?

What metrics are needed? Energy (units of kWh or MWh) for at least the entire uploaded time series, but perhaps also segmented by quarter, month, and/or date. 

What graphical information is needed? Maybe a single graph that shows time series of all uploaded and modeled AC power at the original time resolution. I'd consider more variables, but I don't want to support graphs for every ModelChainResult attribute. Metrics could also be displayed in bar charts.

I imagine we'll have a base template with the vast majority of the text/code, plus templates for each workflow that have minor modifications to the base template. 

I'm less certain about all of the above for the weather adjusted performance ratio use case.

@cwhanse did the user stories contain any info relevant to reports? If so, it didn't make it into the use cases document.

cwhanse, 2020-12-17:
No specific requests of the reports. I may have inserted that step into the use cases in anticipation of being asked about it.

For website users you are thinking of offering a pdf? I'd like to ask Nate, Saraub and Mark what they would find helpful. My guess is that they'll lean towards a document they could attach and forward to their client.  Maybe a preference for bar charts by month rather than a time series graph.
 
Weather adjusted performance ratio is a single scalar value that could be included in the template, or not. There's a function [performance_ratio_nrel](https://github.com/pvlib/pvanalytics/blob/master/pvanalytics/metrics.py) in pvanalytics.  Could be refactored to pull out a helper which accepts the actual and modeled time series and returns the ratio.

wholmgren, 2020-12-17:
We were not anticipating building in a "download as pdf" option. I'm hoping the browser's print to pdf is acceptable because it took a long time to implement generating pdfs on the server for the Arbiter.

Removing the time series graph would greatly cut down the amount of data transferred back to the client. That makes pages load faster and saves cloud compute $ (we have plenty for the DOE period but maybe the long term operator would care).

The report probably should also include the metadata and the selected workflow options (ie I have ghi, dni, dhi, I have air temperature, etc). Package versions too.

ncroft-b4, 2020-12-21:
I don't think offering a pdf would be a high priority in my opinion. My guess is typical use case would be to take data/charts and the user would add it to a pdf they are building themselves. In fact, we would probably only use the base data and make our own graphs for reports from that, thought others might prefer something different. A summary table would be useful probably. I'd say focus on maybe a single summary graph for the interface just so a user can see the analysis ran and completed, and then providing the result data that they can do what they want with.

wholmgren, 2020-12-22:
Notes from our conversation this afternoon:

- No customization - users get all of the analysis that is fit to print
- Tables of total monthly AC energy and avg monthly irradiance, POA global, temperature, and ?. Some or all of these might need to be restricted to "daytime" periods. Sounds like tables are preferable to bar charts. Not discussed, but I'm guessing it would be valuable to download these tables in csv format.
- Did not hear a desire or use case for daily metrics. 
- Time series variables include: AC power, POA global, cell temperature, and ?. 
  - Includes both the modeled and the user-provided data. We decided that comparing DC power to actuals is out of scope, but it's less clear if it's a valid variable for intermediate values. 
  - AC power is only valid for system and inverter levels. POA global and cell temperature are valid at the array level. 
  - There are a handful of ways that we could present the nested data, but I am a big proponent of a new graph for each unit. So, the simplest approach is 1 graph for all AC power (modeled and actuals), 1 graph for all POA, etc. 
  - If feasible, we should load relatively little time series data by default and make the user click a legend item or expand a div to dynamically load data. It would be really cool if we could support any of the `ModelChainResult` attributes, but that is probably too much to ask for. 
  - Not discussed: interactive figures will be created using plotly.
  - It would be nice to include a "download time series" option to get a csv file corresponding to the data in the graphs.
- If I understand correctly, we need to decide on one of the various flavors of performance ratio.
- Maybe include a capacity factor comparison. Would this require a year of data - no more, no less?

alorenzo175, 2021-01-21:
To be a little more concrete so we can get some dashboard components built, one result from a compute job will be a table (available in csv and arrow formats) like 

| | AC Energy | Global Plane of Array Irradiance | Effective Irradiance | Cell Temperature |
|---|---|---|---|---|
| Jan. | 99.87| 400 | 359 | 25|
| ... | | | | |
| Dec. | | | | |

where irradiance and cell temp are the average over all the arrays over the month. Should these averages be "daytime" for all 4 columns? The Arbiter is using a zenith angle cutoff of 87 degrees to separate day and night, so I'd do the same here.

For the timeseries, my plan is to have an AC power series for each inverter, AC power for the system sum, a weather dataframe (poa global, cell temp, effective irradiance columns) for each array, and a day/night (any additional solar position?) series available from the API. The dashboard can decide what to download and put into a plot, and provide a download link to everything.

wholmgren, 2021-01-21:
I think we're focusing on use cases 1a/b for now, but https://github.com/SolarPerformanceInsight/solarperformanceinsight/issues/65#issuecomment-759772231 is relevant for future work.

I don't recall the motivation for the daytime restrictions for O&M. Maybe someone else can articulate that so we understand if it's an essential requirement or a nice to have.

> The Arbiter is using a zenith angle cutoff of 87 degrees to separate day and night, so I'd do the same here.

In addition to the zenith angle, the Arbiter has non-trivial logic to handle interval labels and averages. It's not immediately obvious to me that O&M users would support the same threshold or interval approach.

alorenzo175, 2021-01-22:
> In addition to the zenith angle, the Arbiter has non-trivial logic to handle interval labels and averages. It's not immediately obvious to me that O&M users would support the same threshold or interval approach.

We're requiring intervals < 1hr, assuming left labels, and using the midpoint of the interval for solar position calculations. So if using an hourly interval, the simple cutoff method would discard shoulder hours only if roughly half are "night". That seems reasonable to me.

wholmgren, 2021-02-01:
I'm trying to determine why my local model doesn't match the SPI output and wishing that SPI would give me DC power or energy.

wholmgren, 2021-02-01:
It would also be helpful if I could download json representations of the pvlib objects used in the computations, though I'm skeptical that we can make that happen with reasonable effort.

wholmgren, 2021-02-10:
closing in favor of more specific issues

closed 2021-02-10 by wholmgren