SolarPerformanceInsight/solarperformanceinsight #88: dashboard review issues

wholmgren, 2021-01-15:
Here's a long list of things I encountered while playing with the dev site this morning. Many of these deserve their own issues, and maybe some of them are duplicates

- [ ] tabbing should skip help boxes
- [ ] sure would be nice if we used an elevation look up service
- [ ] remove the default system name. remove the default inverter/array name or change it to an auto-incremented value like "Inverter A" or "Inverter 1"
- [ ] pvwatts loss parameter `Lid` should be `LID`. LID = Light Induced Degradation
- [ ] clarify units throughout dashboard
- [ ] maybe put another save system button at the bottom of the page
- [ ] Display JSON Summary button should be a block with a collapsible arrow. Maybe "Display/Hide JSON Representation"
- [ ] can we round numbers in the inverter database? for example, `C3: -0.014925999999999998` is annoying to read.
- [ ] "duplicate" or "save as" functionality for systems
- [ ] improve api error handling
- [ ] are the pvwatts modules per string and strings options wired up in pvlib?
- [ ] move tracking box to be above modules per string
- [ ] "my weather data file includes" starts with GHI, DNI, DHI and ends with "effective irradiance", while cell temperature starts with "cell temperature" and ends with "...from irradiance, air temperature...". It would be more consistent if the cell temperature options were reversed.
- [ ] guess at column names. e.g. if a column as "time" then use it for the timestamp column. if a column contains "global" or "GHI" then use it for the GHI column.

lboeman, 2021-01-20:
>     * [ ]  clarify units throughout dashboard

For the System specification page, units are mostly found in the help text which are pulled from the API specification. Should the units be displayed next to the inputs as they are in the arbiter? 

>     * [ ]  are the pvwatts modules per string and strings options wired up in pvlib?

Looks like those options were added with the new `Array` class  in https://github.com/pvlib/pvlib-python/pull/1076 ? 

>     * [ ]  guess at column names. e.g. if a column as "time" then use it for the timestamp column. if a column contains "global" or "GHI" then use it for the GHI column.

Seems like guessing at the time column would be simple, but guessing at other columns, especially with different granularity might be a problem.

wholmgren, 2021-01-20:
> For the System specification page, units are mostly found in the help text which are pulled from the API specification. Should the units be displayed next to the inputs as they are in the arbiter?

I missed that. I guess I'm not sure what's best at this point. 

> Seems like guessing at the time column would be simple, but guessing at other columns, especially with different granularity might be a problem.

Yeah, I can see how the other columns are a lot more difficult if there's more than one input per system.

lboeman, 2021-01-21:
In terms of elevation lookup service, I took a quick stab at https://www.opentopodata.org but the appropriate CORS headers are not set on their public API and so it cannot be accessed from the browser One option here is add an endpoint to our API to proxy requests over to them. I'm not sure how involved we want to get with this feature.

alorenzo175, 2021-01-27:
On the units issue, I think we do need to clearly have the expected units of things on the weather upload page e.g. Global Horizontal Irradiance (required, W/m^2). For performance, are we always expecting watts? Do you need to be able to support W, kW, MW?

wholmgren, 2021-03-08:
this is all done, ignorable, or handled in other issues/prs.

closed 2021-03-08 by wholmgren