SolarPerformanceInsight/solarperformanceinsight #98: Add units to upload mapping process

lboeman, 2021-01-28:
Allow users to define what units they have in their data files. Convert to expected base units before API submission. Users should also be able to select units on the results page as well, so standalone conversion functions are a good idea.

alorenzo175, 2021-02-08:
Will definitely need this for the 2abc cases since people are unlikely to start with data in watts

lboeman, 2021-02-22:
Ideally unit conversion code will extend to displaying resulting values and plots in the desired units. Should this also convert values in downloaded data as well?

wholmgren, 2021-02-22:
> Should this also convert values in downloaded data as well?

I think we all agreed this was not necessary and it would be extra work because it would require an API feature instead of just a dashboard feature.

PR #145 (2021-02-25): Allow users to define performance units
closes #98
closed 2021-02-25 by lboeman