SolarPerformanceInsight/solarperformanceinsight #109: update _modelchain_ac_model for multi array inverters

wholmgren, 2021-02-01:
https://github.com/pvlib/pvlib-python/pull/1147 and https://github.com/pvlib/pvlib-python/pull/1150 changed the object oriented api for the new multi-array inverter functions. Now you just pass "sandia" or "pvwatts" and the objects determine which function they should call. So SPI needs to update `PVWattsInverterParameters._modelchain_ac_model` and `SandiaInverterParameters._modelchain_ac_model`. The change is available in `v0.9.0-alpha.2`

alorenzo175, 2021-02-03:
We'll need to wait for https://github.com/pvlib/pvlib-python/pull/1157 otherwise we'll have to adjust the logic below to not return a list if `num_arrays == 1`

https://github.com/SolarPerformanceInsight/solarperformanceinsight/blob/0a7a476d6d4bef99565bd5fe3c44039277f03281/api/solarperformanceinsight_api/compute.py#L89-L111

wholmgren, 2021-02-04:
Merged that PR. Do you want an alpha-3 now?

alorenzo175, 2021-02-04:
I think there is still the issue raised in #119. I'll try and track it down and raise in pvlib

PR #124 (2021-02-08): update for pvlib 0.9.0a4
a4 because https://github.com/pvlib/pvlib-python/pull/1162 is required
along with a tbd issue for computing from effective irradiance

closes #119 closes #109

also adds real tests of the pvlib modelchain to catch errors there
commit 5db53ab2 (2021-02-08): update for pvlib 0.9.0a4 (#124)

* update for pvlib 0.9.0a4

a4 because https://github.com/pvlib/pvlib-python/pull/1162 is required
along with a tbd issue for computing from effective irradiance

closes #119 closes #109

also adds real tests of the pvlib modelchain to catch errors there

* mypy

* test more modelchain configurations

* fix when missing poa global or solar position for from_effective_irr

* make sure poa_global is NaN not None

* use pvlib 0.9.0a4

* comment update
closed 2021-02-08 by alorenzo175