SolarPerformanceInsight/solarperformanceinsight #110: More Dashboard Review Notes

ncroft-b4, 2021-02-02:
While working through a Calculate Predicited/Reference Performance workflow on a new system with a 'pvsyst' model, here are some issues and notes I discovered.

- [ ] Can there be a module lookup like there is an inverter lookup?
- [ ] I found the tags for the module parameters in the system creation confusing. I was trying to use a module from the SAM CEC database, LG395N2W-A5. The tags in SPI (Gamma Ref, Mu Gamma, etc) did not seem consistent with the data in the cec module database that is retrievable in pvlib. In particular, Gamma Ref is a_ref in the database, and Mu Gamma and R SH 0 are not in the database. In the end, the computation on my system did not produce positive energy, possibly to do with some mistake in these parameters that I made.
- [ ] Quality of Life: Would be nice if the timezone automatically filled in from the latitude and longitude. timezonedb.com is one source that can provide that, but it would require maintaining an access key. 
- [ ] Can we provide a note about what format the timezone needs to be in? I used 'Etc/GMT+5" and it worked well for my dataset that did not have a dst shift, but I tried at first just the offset as +5 and got an error, and I am not sure what subset of dst timezones it will accept.
- [ ] I'd suggest leaving the offset off of the date for the start and end of calculations. Have it be a generic date but then map it to the given timezone. I left it at 0, and then the output csvs started with a 5 hour offset from my data start and ended before my input data ended.
- [ ] Can we allow for skipping rows on the input weather csv? Or add a note to make it clear it must be one row of headers followed immediately by data?
- [ ] The process stalled for me a couple times when trying to load in weather data where the format did not match, such as when I had a second header row before the data started. There was not error message or progress indicator, so it was not clear there was an error. Can we add a progress indicator or pass errors on the csv upload and parsing back to the user? I had another error where there was a single "end of dataset" value on the last row of my data that also caused a stall before I realized and fixed it. 
- [ ] I had a two inverter system each at a different orientation with one array. At the end, the graph offered me Array1 above inverter 1, and Array 1 above inverter 2. Both graphs were the same, but when I downloaded the data the csvs were different, as I expected. So there might be an issue with the graph displaying the different arrays. Or I may have needed to name the arrays differently.
- [ ] Can we add an ption to download all the timeseries data in one download?
- [ ] Can we add system-level timeseries output?
- [ ] Quality of Life: Would be nice to have a date picker instead of the date text input on the setup page.

alorenzo175, 2021-02-02:
>   Quality of Life: Would be nice if the timezone automatically filled in from the latitude and longitude. timezonedb.com is one source that can provide that, but it would require maintaining an access key.

#111 

>     Can we provide a note about what format the timezone needs to be in? I used 'Etc/GMT+5" and it worked well for my dataset that did not have a dst shift, but I tried at first just the offset as +5 and got an error, and I am not sure what subset of dst timezones it will accept.

#112 

>     I'd suggest leaving the offset off of the date for the start and end of calculations. Have it be a generic date but then map it to the given timezone. I left it at 0, and then the output csvs started with a 5 hour offset from my data start and ended before my input data ended.

#113 Tooltips showing the descriptions from https://github.com/SolarPerformanceInsight/solarperformanceinsight/blob/0a7a476d6d4bef99565bd5fe3c44039277f03281/api/solarperformanceinsight_api/models.py#L575-L601 should help

alorenzo175, 2021-02-02:
> Can there be a module lookup like there is an inverter lookup?
>
> I found the tags for the module parameters in the system creation confusing. I was trying to use a module from the SAM CEC database, [...]

I think the problem here is that the CEC module parameters are subtly different from the pvsyst module parameters, so we can't just use the CEC module database.

cwhanse, 2021-02-02:
> I think the problem here is that the CEC module parameters are subtly different from the pvsyst module parameters, so we can't just use the CEC module database.

That is correct. The idea behind the pvsyst model configuration is that a user has the parameter values from a PAN file.  We could open an issue to implement a CEC model configuration, we discussed it before but pushed it off as a nice-to-have.

ncroft-b4, 2021-02-02:
> > Can we add system-level timeseries output?
> 
> What do you mean by this? There should be timeseries output of the performance data in the dropdown (like "My System performance data"). Since arrays can have different orientations, etc. weather data is only per array.

Yup thanks, I see that now, I missed it. I do think it would be handy if that could include the same combining of weather data into one system level stream that is done for the monthly table, but as the time series output, but its not necessary I don't think.

ncroft-b4, 2021-02-02:
> That is correct. The idea behind the pvsyst model configuration is that a user has the parameter values from a PAN file. We could open an issue to implement a CEC model configuration, we discussed [...]

It is too bad, because that does feel a bit at odds with the intended audience, which would be for users who might not be so savvy as to find those attributes in a pan file. I struggled with the translation. But understand that it is a current limitation. What about a inverter and module lookup for pvwatts models? That would provide a model that is a very straightforward entry point for users who might know what equipment they have but don't have all the parameter data for it. But I also don't mean to bog up too much time on this point.

wholmgren, 2021-03-08:
the last items are tracked in issues #154 (addressed in #162) and #135 (addressed in #149), so closing this for simplicity.

closed 2021-03-08 by wholmgren