SolarPerformanceInsight/solarperformanceinsight #154: improve date/time picker

wholmgren, 2021-02-26:
The date picker looks nice and works well for recent dates, but it is extremely tedious to specify a start and end date like jan 1, 2019 through dec 31, 2019. 

It's also not clear that the time picker will come after the date picker, so I was surprised as a first time user. Is it possible to set the default start time to 00:00 and the default end time to 23:59?

Maybe a menu based system would be easier. Some users could still type the date if they tab between fields and take advantage of browser autoselection, while users could still point and click.

ncroft-b4, 2021-02-26:
Should the end time be the last minute of the period covered (i.e. 23:59), or the time on the last timestamp? I tried to upload a file and the last timestamp was 12/31/1990 23:00 and so I was inputting 23:00 for the time. I also found the time input to be the most tedious part of the process.

I was also concerned/confused about the time zone still. My data was in America/New_York, but the time zone was not indicated in the timestamps. So the start time for SPI was 01-01-1990 00:00-5:00 but on the data it was 1/1/1990 0:00 . Not sure if that would be seen as a problem or if it would map the America/New_York timezone to my input timestamps.

wholmgren, 2021-02-26:
> Should the end time be the last minute of the period covered (i.e. 23:59), or the time on the last timestamp? I tried to upload a file and the last timestamp was 12/31/1990 23:00 and so I was inputting 23:00 for the time. 

I don't know the answer of the top of my head, so we definitely need to document it right there. 

> I was also concerned/confused about the time zone still. My data was in America/New_York, but the time zone was not indicated in the timestamps. So the start time for SPI was 01-01-1990 00:00-5:00 [...]

Hm, I thought were were going to reject timestamps that are not localized. We should certainly make it clear how the timezone will be used.

`America/New_York` includes a DST shift. Does your data include that shift as well? In any case, I'm wondering if we need to say something about using the Etc/GMT timezones.

ncroft-b4, 2021-02-26:
> America/New_York includes a DST shift. Does your data include that shift as well? In any case, I'm wondering if we need to say something about using the Etc/GMT timezones.

Good point. My data did not include a time shift, so it really was Etc/GMT+5, and I was even trying to input it wrong. I will never not get frustrated dealing with time zones in data. The point still holds I suppose. 

We could do a bit more handholding through the process perhaps? Start with the assumption the data is in the timezone of the location of the site, whether it is localized or not. Give the user a checkbox for if it has a dst shift or not. Then the tool forces a timezone on it from the user input for **local** start and stop time with out time zone data. Might something like that work?

I am not clear exactly how/why the tool takes the user input dates and then maps it to the dates in the data. Maybe something to discuss and then try to sort out the most user friendly workflow that is possible.

alorenzo175, 2021-03-01:
> Should the end time be the last minute of the period covered (i.e. 23:59), or the time on the last timestamp? I tried to upload a file and the last timestamp was 12/31/1990 23:00 and so I was inputting 23:00 for the time. I also found the time input to be the most tedious part of the process.

23:59 or even 1/1/2000 00:00 is acceptable for the end. Definitely need a tooltip or even text right there for clarity

> I was also concerned/confused about the time zone still. My data was in America/New_York, but the time zone was not indicated in the timestamps. So the start time for SPI was 01-01-1990 00:00-5:00 [...]

Another thing to clarify for sure. The timezone you select is the timezone all the calculations and aggregations are done in. If the timezone is left off either start/end or data uploads, it is assumed to be localized to the job timezone.

> We could do a bit more handholding through the process perhaps? Start with the assumption the data is in the timezone of the location of the site, whether it is localized or not. Give the user a [...]

Yes, if we get #111 implemented, we could give an initial timezone suggestion for a job. In that case I guess it isn't so bad to have a checkbox for dst or not. Also not bad if it's a standard timezone like America/New_York to convert to Etc/GMT+5, but then there are corner cases like America/Phoenix. I guess something like a "Suggested" section at the beginning of the timezone list could include recommendations like America/New_York or Etc/GMT+5 but you could still scroll down and choose any timezone.

> I am not clear exactly how/why the tool takes the user input dates and then maps it to the dates in the data. Maybe something to discuss and then try to sort out the most user friendly workflow that is possible.

The reason the start/end/step/timezone is required when you're just starting a job is to allow strong data checks when you upload data. It allows the user to immediately get feedback on if the data is formatted correctly.  For example, the API returns a good bit of information about missing or extra data in each upload https://dev.solarperformanceinsight.org/api/docs#/Jobs/post_job_data_jobs__job_id__data__data_id__post that could be exposed in the dashboard. 

This whole initial validation is especially important when you're uploading weather and performance for the use cases 2a-d. I know I'd be annoyed if I uploaded weather data and performance data that were misaligned, but was able to go through and click compute only to find out (maybe minutes once server is loaded) later that the job failed because my data was off. Or worse, that the results are wrong because of some data issue. I do agree that it can be overly burdensome when just starting a job and possibly drive users away in frustration.

PR #162 (2021-03-09): Improve timefield
closes #154
closed 2021-03-09 by lboeman