SolarPerformanceInsight/solarperformanceinsight #166: Dashboard Support for Usecase 2A/2B

lboeman, 2021-03-08:
After the call Monday, it's clear that some data restrictions will need to move levels to support 2A and 2B. We had discussed how to make the dashboard form more simple for the compare performance use cases, but we may have oversimplified. I think the restrictions we have on the data are important in ensuring that we can proceed with the computation without any gotchas or incompatibilities that only crop up once you try and run the comparison. I think we can support 2A/B while still avoiding those issues by making a form like 
[image]

Then we would be able to support different weather formats for the two uploads, and different time indices. We would add some validation to make sure the predicted time index could be shifted to the actual/expected time index.

_Originally posted by @alorenzo175 in https://github.com/SolarPerformanceInsight/solarperformanceinsight/issues/65#issuecomment-768446066_

lboeman, 2021-03-08:
Probably best to create a component to handle the `PredictedDataParams` class as well as the `ActualDataParams` and a forthcoming `ExpectedDataParams` @alorenzo175 is it safe to assume the parameters for expected and actual parameters are going to be the same?

alorenzo175, 2021-03-08:
No there are slight differences. They are actually already implemented. Probably easiest to use the API docs as a guide. ComparePredictedActualJobParameters

Some differences from the screen shot include only a single set of time parameters like all the other jobs and predicted data params has the additional `data_available` param

alorenzo175, 2021-03-08:
For 2B/`ExpectedDataParams` I think it's probably ok to assume expected and actual parameters will be almost the same. The 'I will provide performance data..'/performance granularity should be left off though

PR #170 (2021-03-15): Usecase 2A dashboard implementation 
closes #166
closed 2021-03-15 by lboeman