SolarPerformanceInsight/solarperformanceinsight #180: compare use case feedback

wholmgren, 2021-03-16:
@cwhanse @ncroft-b4 fyi use case 2A is ready for testing. 

My first round of notes below...

Buttons appear to show/hide options as expected - nice job with that complexity. 

[image]

In the time parameters description, change "An attempt will be made to" to "SPI will attempt to".

Timezone help text: "If your data does not account for daylight savings, use the appropriate fixed-offset. For example, to specify year-round Mountain Standard Time, select Etc/GMT+7"

Start/end boxes might look better if they used capitalized letters: YYYY:MM:DD

Timebox error message: "you specified 60 (of type number) as a minute, which is invalid"  - can we drop the "(of type number)"?

Time widgets are much improved overall.

[image]

I'll generate some data and run through the process later today.

lboeman, 2021-03-16:
The reason for the grey box width is to fit the large monthly data labels. Kind of a pain to get that sidebar to behave reasonably and dynamically size.
[image]
 
> Timebox error message: "you specified 60 (of type number) as a minute, which is invalid" - can we drop the "(of type number)"?

I'm just printing the error messages from luxon here, I'll make a quick pass at stripping it out.

wholmgren, 2021-03-16:
I set the wrong year in my simulated data and tried to upload it. I got this error message:

> Over 100.00% of rows were missing from the data upload to conform to the job's stated time index.

is there a reason we include "over" in this message? Also I think we should round to the nearest integer percent.

wholmgren, 2021-03-16:
I'm not sure what I'm supposed to do with the information in this table

[image]

I already provided the data, so I'm not interested in downloading it again. The table doesn't retain information about my filenames, so I can't check that I mapped the right file to the right system component. I had to provide the correct variables in order to get to this point. It's not obvious if I'm able to click back to change something or if that would corrupt the job and make me start over. 

So again my feeling is that this page just slows me down and results should automatically be calculated after the last submission.

wholmgren, 2021-03-17:
@cwhanse should we change names as originally discussed in https://github.com/SolarPerformanceInsight/solarperformanceinsight/issues/65#issuecomment-759739523?

> I regret that we kept the terms "predicted, expected" in the use case descriptions. I think these terms have only obscured the intent and made designing the interface difficult. In hindsight, the terms actual, modeled (in place of expected) and reference (in place of predicted) would have been more clear, at least to me.

cwhanse, 2021-03-17:
change terms on the dashboard pages? I vote yes. @ncroft-b4 and I need to draft the user guide and it is my intent to use "actual", "modeled" and "reference" rather than "actual", "expected" and "predicted". We'll probably cross-reference "modeled" to "expected" for people more familiar with the NREL lingo.

wholmgren, 2021-03-17:
[image]

wholmgren, 2021-03-17:
Can we remove "performance data" from the legend? Also I think it we need to clarify that it's weather adjusted predicted

[image]

ncroft-b4, 2021-03-25:
I would like to be able to plot the system-wide results that come from the compare workflow. If I upload site level actual performance, I will want to compare to site level weather adjusted, but the only options are inverter level. 

[image]

I can add all the inverters but that does not make for a very useful graph. 

[image]

And I could download the inverter level results and sum them myself and make my own graph, but I would like the tool to be able to do that.

ncroft-b4, 2021-03-25:
When uploading actual power data per inverter, the prompts do not let me know which inverter (or is it meant to be array?) I am specifying data for. I assume it is in order of the inverters that were setup in the system but it would be handy to have it be labeled as such.

[image]

ncroft-b4, 2021-03-25:
There are a couple quirks in the use cases in the compare case between uploading actual power on a site-level vs inverter level. 

If I upload site level, I am really only interested in getting site-level weather adjusted in return, as mentioned above.

If I upload inverter-level, I can't also upload site level. So that limits the use case there to just being inverter level comparisons. That all works, and the results are very useful for that. However, without the site-level meter data I don't need the site-level comparisons that the tool returns. So the Monthly Summary results table and results download become kind of extraneous, because I am going to want to have those with meter data, not inverter sums in most cases. 

This is all just quality of life. But ideally, I could upload both the site level actual power and inverter level power at the same time. As it stands now I need to do the comparison workflow twice, once to get useful site-level results and once to get useful inverter level results. 

It might not be worth the coding time to fix this. As a user I could always just run the inverter level version of the tool, download the results, and then on my end just use my meter data to make the site-level comparisons. But I do think it would be compelling to eventually be able to do that all in one go in SPI.

closed 2021-03-25 by lboeman

lboeman, 2021-03-25:
> When uploading actual power data per inverter, the prompts do not let me know which inverter (or is it meant to be array?) I am specifying data for. I assume it is in order of the inverters that were setup in the system but it would be handy to have it be labeled as such.
> 
> [image]

This should be fixed with the merging of #184. And there are some other minor updates, the most notable of which is no more "submit" step. When all data is uploaded you'll be kicked over to the results page and the job will automatically compute.

PR #184 (2021-03-25): 2a feedback updates
PR for handling smaller issues raised in #180. 

closes #180, other items in follow up issues.