SolarPerformanceInsight/solarperformanceinsight #190: Correct calculation of temperature factor

cwhanse, 2021-03-23:
The use case descriptions I provided assumed that the temperature coefficient `gamma` was a positive number. That assumption conflicts with pvlib-python's conventions. [This calculation](https://github.com/SolarPerformanceInsight/solarperformanceinsight/blob/f75f54e4ca45bed31925f7cd977bd809a76a2245/api/solarperformanceinsight_api/compute.py#L514) should change to

```
return _zero_div(1 + gamma * (t_actual - t0), 1 + gamma * (t_ref - t0))
```

Is there a numerical test of `compare_predicted_and_actual` that checks correctness of the adjusted energy? Or should I come up with one?

wholmgren, 2021-03-23:
Good catch.

I'm not very familiar with testing `compute.py` because of the heavy dependencies. I thought this test would fail if those signs are changed but it does not.

https://github.com/SolarPerformanceInsight/solarperformanceinsight/blob/f75f54e4ca45bed31925f7cd977bd809a76a2245/api/solarperformanceinsight_api/tests/test_compute.py#L785-L816

wholmgren, 2021-03-23:
Ah, I forgot the monthly temperature correction is different. I think the `-` in this line should also be changed to a `+`:

https://github.com/SolarPerformanceInsight/solarperformanceinsight/blob/f75f54e4ca45bed31925f7cd977bd809a76a2245/api/solarperformanceinsight_api/compute.py#L816

The test in the comment above does fail when I apply that change.

wholmgren, 2021-03-23:
This test coarsely checks the correctness of the corrected energy

https://github.com/SolarPerformanceInsight/solarperformanceinsight/blob/f75f54e4ca45bed31925f7cd977bd809a76a2245/api/solarperformanceinsight_api/tests/test_compute.py#L1031-L1073

There are 270 permutations run within that test. Perhaps we need a more precise test for a single permutation.

PR #191 (2021-03-24): fix temperature loss coefficient
closes #190
closed 2021-03-24 by wholmgren